ceva-dsp/sh2-demo-nucleo #2: GetProductID Problem

ldelmont, 2019-08-19:
I have a problem to obtain de Product ID, when i call the function getProdIdStart the value of the channel is still 0xFF. I don't understand what is wrong or what i have to modify ?

closed 2019-08-19 by ldelmont

dave20874, 2019-08-19:
When you call sh2_open(), does it return 0 (SH2_OK)?  One of the tasks of open() is to establish communications with the chip and set the channel from 0xFF to the real value.

My guess is that communications between BNO080 and host are not working and sh2_open() is failing.  You’ll need to address the problem in your SH2 HAL implementation.

Dave Wheeler

ldelmont, 2019-08-20:
Yes, it return SH2_OK. I am using the same function of the git project but with stm32l010rb microcontroller. Now , I am thinking that the problem is with the timer, because I have a 16 bits timer. I have modified many constants values with respect delays but still I have not found a correct way.

Lucas Adrian Delmonte
Junior Engineer

Energy & Engineering Business Unit
Smart Systems
ldelmonte@leitat.org

[cid:image002.png@01D01AB2.E15D18A0]

C/ de la Innovació, 2<http://www.leitat.org/castellano/leitat/leitat6.htm> ·
08225 Terrassa (Barcelona)
Tel. (+34) 93 788 23 00 ·
Fax (+34) 93 789 19 06
www.leitat.org<http://www.leitat.org/>

Enviado el: Monday, August 19, 2019 4:06 PM
Para: hcrest/sh2-demo-nucleo <sh2-demo-nucleo@noreply.github.com>
CC: Lucas Adrián Delmonte <ldelmonte@leitat.org>; State change <state_change@noreply.github.com>

ldelmont, 2019-08-20:
Hi Dave , now the channel is correct but when sh2_getProdIds is call  it doesn't go out of this loop of the function opProcess(pSh2, &getProdIdOp);

> `// While op not complete and not timed out.

    while ((pSh2->pOp != 0) &&
           ((pOp->timeout_us == 0) ||
            ((now_us-start_us) < pOp->timeout_us))) {
        // Service SHTP to poll the device.
        shtp_service(pSh2->pShtp);

        // Update the time
        now_us = pSh2->pHal->getTimeUs(pSh2->pHal);
}`

Do you have any idea of the problem?  or what i should check ?

Thanks, Lucas

kzger, 2022-09-20:
> Hi Dave , now the channel is correct but when sh2_getProdIds is call it doesn't go out of this loop of the function opProcess(pSh2, &getProdIdOp);
>
>
> ```
> while ((pSh2->pOp != 0) && [...]

Hi, @ldelmont and @dave20874.

Have you solved the problem? 

I have same issue too.

dwheeler-ceva, 2022-09-20:
@kzger,

When the code is in this loop, the host is waiting for a response from the device.  If you're stuck here, that usually means basic communications are not working.  My advice would be to connect a logic analyzer to the I2C, SPI or UART lines to see what's happening.  You may also want to probe the interrupt line and check the states of the reset, PS0, and PS1 lines.  If you can't do that or you're stuck, can you provide more details on the type of interface and host you're using?

Dave

kzger, 2022-09-21:
@dwheeler-ceva 

Thanks for responding.

I am using stm32F72x series to communicate bno080 via SPI1.

Already tried to replace interrupt pin with gpio_input pin, and read that pin with GPIO_ReadPin inside main loop, unfortunately, still no luck.

dwheeler-ceva, 2022-09-21:
Do you have a logic analyzer, like a Saleae Logic unit, that you can use to see the SPI bus?  Some things to check: Is the WAKEN signal being asserted to the bno080?  Is the INTN signal being asserted by by bno080 to the device?  Is CSN being asserted to the device before the SPI transaction?    Do the SPI clock and data lines toggle as expected?

kzger, 2022-09-23:
@dwheeler-ceva

Thanks for your advice.

The problem have been solved by increase more delay time to wait bno080 to respond.