ceva-dsp/sh2-demo-nucleo #3: Synchronization

ldelmont, 2019-09-23:
I have activated SH2_ACCELEROMETER, SH2_GYROSCOPE_CALIBRATED and SH2_MAGNETIC_FIELD_CALIBRATED at the same report Interval. i am receiving almost all the time Accelerometer data. Every 5 or 6 accelerometer data i am receiving one gyroscope or mangnetomer data.

E.g: Acc, Acc, Acc, Mag, Acc, Acc, Acc ,Acc, Gyr, Acc, Acc ,Acc,

 What should i do to receive one data of each sensor ordered.
E.g: Acc, Mag, Gyros, Acc, Mag, Gyros, Acc, Mag, Gyros

Thank you.

dave20874, 2019-09-23:
What report interval did you request?  The sensors only support certain discrete rates.  When you request an arbitrary interval, the sensor is configured for the best supported rate.  This is usually the next higher rate than what you requested or the highest available rate.

The problem could also be that your communications channel can't support the number of sensor reports requested.  This may happen when using i2c, for example, and enabling several sensors at a high rate.  The I2C bus simply can't carry all the requested reports and some are dropped.

ldelmont, 2019-09-24:
Yes i am using i2c but the request is every 100ms for the 3 sensors.

ldelmont, 2019-09-24:
With the same code  sh2-demo-nucleo of the repository I have the same problem.

Configuration:
[image]

Terminal :
[image]

Interruptions: sometimes I have 100 ms between the same sensor interrupt and sometimes no.
[image]

In the case of my personal project I think that the result is worst because I am working with a cortex m0 (stm32l010).