ceva-dsp/sh2-demo-nucleo #5: Will this project run on SW4STM32 ?

Ajay-Embed, 2019-11-18:


dave20874, 2019-11-18:
It is set up for IAR Embedded Workbench for ARM, which is available with a free license for small projects.  To work with SW4STM32 you would have to set up a project in that IDE but all the source code should work without modification.

Ajay-Embed, 2019-11-18:
Thankyou for the reply Dave, I was going through the source code and I must say this is a brilliant project considering how hard it is actually to obtain game rotation vectors from an industrial grade sensor such as BNO080, also does the code print values on a serial software (eg. teraterm)  or on the debug console?
Thanks again!

dave20874, 2019-11-18:
The file app/console.c implements the stdout functions.  As provided, it writes to USART2, which is transported over the Nucleo's USB interface and appears as a virtual com port on the host system.  So you can use Teraterm or other serial software to talk to it.  

At times, I've tried using the debug console but found it interferes with real-time processing on the target system.

Ajay-Embed, 2019-11-19:
Thanks for the info Dave.

closed 2019-11-19 by Ajay-Embed