ceva-dsp/sh2-demo-nucleo #6: Report Interval with Gyroscope or Rotation Vector

Malo448, 2019-11-28:
Hello,

I have an issue, when I set "config.reportInterval_us" lower than 5000 (200Hz) but greater than 2500 (400Hz), for example : 4000 (250Hz).

When a put this configuration, I have a report at approximatively 400Hz with the rotation vector or the gyroscope enabled whereas I asked for report data at the rate of 250Hz. But when the accelerometer is enabled, I don't have this problem and I have accelerometer data report at 250Hz like I want it.

Do you know what can be the problem and how to resolve it please ?

Thank you in advance for your answer.

Best regards

Malo

dave20874, 2019-12-02:
The accel and gyro sensors only work at specific, discrete rates.  Whatever rate you request, you will get the next higher rate that is actually supported.  To know the actual rate that was adopted, you can read back the sensor's configuration.  Unfortunately, the gyro and accelerometer support different rates, too, so asking for these two with the same rate can produce data streams with different rates.  The fused sensors, like Rotation Vector, generally follow the rates of the gyro.

Malo448, 2019-12-02:
Hi Dave,
Thank you, that what we were thinking.
Best regards,
Malo

closed 2019-12-02 by Malo448