ceva-dsp/sh2-demo-nucleo #12: in file i2c_hal.c the READ_LEN is wrong

yao2015lin, 2022-06-20:
in file i2c_hal.c the READ_LEN is wrong

yao2015lin, 2022-06-20:
the correct value is 4

dave20874, 2022-06-20:
Why do you say it should be 4?  The i2c HAL is designed to read incoming data with 2 i2c transactions: the first reads just two bytes and these contain the length field of the header.  Then, knowing the amount of available data, it performs a second read with the full length.  The data in the second read begins with another header so, in effect, the length field is read a second time.  We want to minimize the amount of redundancy between these two read operations, though, so we've always used 2 as the read length for that first transaction.